Ro5bert/avra #2: Support for ATmega644

5inf, 2019-07-20:
When trying to assemble the following code https://github.com/5inf/uprog2/tree/master/source/PROG-USB
avra complains about the missing m644.inc.
even with a m644.inc (e.g. from here https://github.com/DarkSector/AVR/blob/master/asm/include/m644def.inc) 
avra throws the error "m644def.inc(47) : Error   : Unknown device: ATmega644"

```
$avra main-usb.asm
[…]
Pass 1...
m644def.inc(44) : PRAGMA directives currently ignored
m644def.inc(47) : Error   : Unknown device: ATmega644
m644def.inc(48) : PRAGMA directives currently ignored
m644def.inc(53) : PRAGMA directives currently ignored
m644def.inc(1018) : PRAGMA directives currently ignored
m644def.inc(1019) : PRAGMA directives currently ignored
m644def.inc(1020) : PRAGMA directives currently ignored
m644def.inc(1021) : PRAGMA directives currently ignored
main-usb.asm(129) : Error   : Cannot find include file: devices_no_public/zspm.asm
Warning : No .DEVICE definition found. Cannot make useful address range check !
```
What is needed to add support for the ATmega644?

hsoft, 2019-07-20:
An entry in https://github.com/hsoft/avra/blob/master/src/device.c#L47

5inf, 2019-07-20:
Ok, I found the list but did not dare to think it was that simple. It seems to be working. You have a pull request. Thanks!

closed 2019-07-20 by 5inf

smbraden, 2020-11-25:
I have a similar error, but with an ATmega328, which is listed. Any thoughts about how to work out my error?

`Pass 1...
m328def.inc(44) : PRAGMA directives currently ignored
m328def.inc(47) : Error   : Unknown device: ATmega328
m328def.inc(48) : PRAGMA directives currently ignored
m328def.inc(53) : PRAGMA directives currently ignored
m328def.inc(922) : PRAGMA directives currently ignored
m328def.inc(923) : PRAGMA directives currently ignored
m328def.inc(924) : PRAGMA directives currently ignored
m328def.inc(925) : PRAGMA directives currently ignored
GNU-AVR.asm(18) : Error   : Unknown device: ATmega328
GNU-AVR.asm(46) : Error   : Unknown mnemonic/macro: delay
Warning : No .DEVICE definition found. Cannot make useful address range check !`

Ro5bert, 2020-11-25:
@smbraden What version of AVRA are you using? (Check with `avra --version`.) I ask because, if you happen to be using the version from Debian, it won't have ATmega328 support.

smbraden, 2020-11-25:
`AVRA: advanced AVR macro assembler Version 1.3.0 Build 1 (8 May 2010)
`

Ro5bert, 2020-11-25:
@smbraden Yeah, that version doesn't have ATmega328 support. To get a version with support for ATmega328, you can either build the latest version of AVRA from source (see the README; it is pretty easy), or use a packaged version if it is recent enough (>= 1.4.2).

smbraden, 2020-11-26:
Thanks, got it up and running now. Had just used the apt install before