Ro5bert/avra #21: cross-compiling for mingw32 is failing

A-Ibrahim, 2020-07-25:
I'm trying to compile it for windows but getting this error 
make OS=mingw32
make[1]: Entering directory '/home/pi/avra/src'
rm -f avra *.o *.p *~
make[1]: Leaving directory '/home/pi/avra/src'
make -C src -f makefiles/Makefile.mingw32
make[1]: Entering directory '/home/pi/avra/src'
x86_64-w64-mingw32-gcc -Wall -O3   -c -o avra.o avra.c
x86_64-w64-mingw32-gcc -Wall -O3   -c -o device.o device.c
x86_64-w64-mingw32-gcc -Wall -O3   -c -o parser.o parser.c
x86_64-w64-mingw32-gcc -Wall -O3   -c -o expr.o expr.c
x86_64-w64-mingw32-gcc -Wall -O3   -c -o mnemonic.o mnemonic.c
x86_64-w64-mingw32-gcc -Wall -O3   -c -o directiv.o directiv.c
x86_64-w64-mingw32-gcc -Wall -O3   -c -o macro.o macro.c
macro.c: In function ‘expand_macro’:
macro.c:502:5: warning: ‘strncat’ specified bound 1 equals source length [-Wstringop-overflow=]
     strncat(buff, "\n", 1);
     ^~~~~~~~~~~~~~~~~~~~~~
x86_64-w64-mingw32-gcc -Wall -O3   -c -o file.o file.c
file.c: In function ‘open_out_files’:
file.c:98:11: error: ‘VERSION’ undeclared (first use in this function)
           VERSION, basename, ctime(&pi->time));
           ^~~~~~~
file.c:98:11: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [<builtin>: file.o] Error 1
make[1]: Leaving directory '/home/pi/avra/src'
make: *** [Makefile:23: all] Error 2

Ro5bert, 2020-07-25:
Hi,

Thanks for reporting this issue. It looks like the makefile for mingw32 is rather out-of-date. In particular, it's not including `-DVERSION` as a compiler argument, hence the error. I've updated `Makefile.mingw32` to work more like the makefile for Linux, which is known to work well. Unfortunately, I do not have a Windows machine to test it on. Do you think you could try recompiling with the change to the makefile and let me know if it works? I've put the changes in the [issue-21 branch](https://github.com/Ro5bert/avra/tree/issue-21).

Robert

A-Ibrahim, 2020-07-25:
I pulled the new changes that you made on makefile.mingw32 for 'issue-21' branch and compiled again, it worked perfectly
FYI: I'm using raspberry pi w for cross compilation

Ro5bert, 2020-07-26:
Good to hear. I will merge in the change when I get home later.

commit 76f16b9e (2020-07-26): Merge branch 'issue-21'

Resolves #21
closed 2020-07-26 by Ro5bert