Ro5bert/avra #24: Update Apt repository version

Stateford, 2020-10-12:
I attempted to use the apt-get repo for `avra` and there were missing includes.

`m328Pdef.inc` to be specific. I had to uninstall and clone this repo to get a package that included the correct file. I think it would be a good idea to consider updating the ubuntu repo's if possible.

Ro5bert, 2020-10-13:
I wasn't even aware it was on Ubuntu's repo. Thanks, for letting me know. I'll try to update it.

bwack, 2020-10-23:
I second this. Great idea to update the apt repo. I'm on ubuntu 20, and the avra version I have is from 2010. The -o output file name option has no effect. I was compiling for two targets in one makefile so I had to use the move command, until I found that avra was on github :) 
Great project, thanks for keeping it alive !

mase76, 2020-10-30:
The same issue in Debian.

Ro5bert, 2020-11-01:
@mase76 Yeah, the Ubuntu package is inherited from Debian. There's actually an open [bug](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=969580) from 5 September about AVRA on Debian (and, hence, distros derived from Debian) being outdated and still using the old Sourceforge upstream, but the maintainer hasn't responded. I'm new to Debian packaging, so I'm not sure exactly what the procedure is for dealing with unresponsive maintainers. However, there's a [wiki page](https://wiki.debian.org/qa.debian.org/MIATeam) suggesting that it may take multiple months of trying to contact the old maintainer before they are removed/the package is handed off or orphaned.

mase76, 2020-11-01:
I think, for a good packaging, the project needs a build system like cmake or similar. Otherwise the prefix path in the Makefile has to be patched. There is also no dependency check.

Ro5bert, 2020-11-01:
I am reluctant to use CMake, but from what I've read, it should be fairly easy to set up the debian package so it sets the prefix to /usr instead of /usr/local when building, so I don't think that's a big issue. As for dependencies, there aren't any besides libc and unistd.h (which does pose a problem from non-POSIX systems, but a workaround has been addressed [here](https://gist.github.com/hack-tramp/b19b7675670bb5463bb763c602b5bc05)).

majekw, 2020-11-01:
@mase76 , there is no need to patch prefix since about year. Look for Slackware packaging script for example: https://slackbuilds.org/slackbuilds/14.2/development/avra/avra.SlackBuild

Ro5bert, 2021-01-27:
I suppose I should update this, seeing as quite a few people are/were interested. I did email the old maintainer of the Debian package a while back and got no response, so I guess it's safe to say he's no longer interested in maintaining the Debian package and someone else should maintain it. However, I started working in December on AVRA 2.0, which is a rewrite of certain parts (most) of the internals (they are currently messy and full of bugs), and I don't think I'll be pursuing updating the Debian package until that's finished. If somebody else with experience in Debian packaging wanted to go through the process of trying to update the Debian package, that'd be great.