Ro5bert/avra #26: Improper warning on inversion operator

Vov4ick, 2020-12-04:
When immediate instruction (ldi, andi, etc) has second operand with inversion operator and MSB of its expression is set, for example, **~(128)**, this produces _constant out of range_ warning. Result of assembly is correct.

Ro5bert, 2020-12-04:
I'll look into this more later, but I'm guessing it's because the internal representation of integers is more than 8bits, so when 128 gets inverted, the result isn't 127. For now you could silence the warning by doing `low(~128)`.

Ro5bert, 2021-01-04:
There's a comment related to this in the [usage file](https://github.com/Ro5bert/avra/blob/1b1b3fb0a3eddc5a3784ad301e3e060f6cb112de/USAGE.md):
> The expression (~0x80) is a Bitwise Not operation. This operator returns the input expression with all its bits inverted. If 0x80  represents -128, then 0x7f, or +127 should be ok. If this is considered as a 32-bit expression (AVRA internal representation), then it appears to be more like oxffffffff-0x80 or 0xffffffff^0x80. The result would then be 0xffffff7f. The assembler would then have to be told or it would have to decide, based on context, how much significance to assign to the higher bits. I have also encountered such conditions with various assemblers, including AVRA. To make sure the assembler does what I really want, I use a construct like 0xff-0x80 or 0xff^0x80. This way the bit significance cannot extend beyond bit-7 and there cannot be any misunderstanding.