Ro5bert/avra #27: .undef not work (or misused by me)

bdyssh, 2020-12-30:
Hello. I need to use `.undef` , but it just does not have any effect. 
How to reproduce: Try to translate three-lines asm code:
```
.define test
.undef test
.define test
```
Current result: I get error:
`undef.asm(3) : Error   : Can't redefine constant test, use .SET instead`
Expected result: No errors.
Versions: This problem exist with old avra 1.3.0 from 2010, as well as latest version 1.4.2.
Am i misuse the directive?
Thanks.

Ro5bert, 2020-12-30:
You're not misusing it; `.undef` is currently not implemented (i.e., AVRA recognizes it, but does nothing). I'll look into providing an implementation for it.

jameswilddev, 2021-04-13:
Will this work like AVR Assembler's undef (http://ww1.microchip.com/downloads/en/devicedoc/40001917a.pdf) or C's undef (https://gcc.gnu.org/onlinedocs/cpp/Undefining-and-Redefining-Macros.html)?  (will it allow reuse of a register, or deletion of preprocesor values)?

Ro5bert, 2021-04-13:
@jameswilddev I planning on making it undefine any symbol (i.e., a combination of the preprocessor and assembler versions of the directive). Sometime before I started maintaining AVRA, the unfortunate decision was made to make the `#directive` syntax merely alias the `.directive` syntax, which has caused many problems, and I can't really change it for backwards compatibility.

Ro5bert, 2021-04-13:
@jameswilddev I planning on making it undefine any symbol (i.e., a combination of the preprocessor and assembler versions of the directive). Sometime before I started maintaining AVRA, the unfortunate decision was made to make the `#directive` syntax merely alias the `.directive` syntax, which has caused many problems, and I can't really change it for backwards compatibility. Earlier this year I started working on AVRA 2.0, which is mostly a rewrite of the internals, but it will also incorporate this change. I've been too busy to work on it recently, but I should be able to this summer. Until then, I'm not fixing any non-major AVRA 1.x bugs, because the code is too messy. Every time I fix a bug, I discover 3 more just reading the surrounding code; it's that bad.

jameswilddev, 2021-04-14:
Aha, I know that feeling.  I'm on and off playing around with a fairly large AVR assembler codebase at the minute so if you need something big to smoke test it with just let me know!

On the other hand I've also recently played around with writing an AVR opcode interpreter as part of a tool to brute force search for solutions to awkward bitwise manipulation which needs to fit into small spaces in terms of flash and cycle count.  So if you need a hand with AVRA 2.0 let me know!