Ro5bert/avra #28: Forward references never allowed in expressions

Ro5bert, 2021-01-05:
Forward references should be allowed in expressions

- as an argument to a mnemonic or macro, and
- as an argument to any directive except conditionals, `.byte`, `.csegsize`, and `.org` (since these directives effect the IP).

Currently they are never allowed in expressions; for example, the following fails:
```
.device atmega2560 ; Arbitrary
.equ myequ=mylab
mylab: mov r1, r2
```

Nevada317, 2021-01-07:
In fact, i think those were allowed back in 1.3.0. Need to check, but looks like they were ruined with ifdef refactor.
I will try to reproduce in older avra soon

Ro5bert, 2021-01-07:
@Nevada317 Actually I realized they do work as arguments to mnemonics, if that's what you mean. It's just as arguments to directives that they don't work, because we try to evaluate expressions on directives in pass 1, even if we don't need to.