Ro5bert/avra #29: Missing include files for some devices

aerobaticant, 2021-02-23:
The includes folder is missing the definition include files for at least the following devices:
ATmega164A, ATmega164PA, ATmega324A, ATmega324PA, ATmega644PA, ATmega1284, ATmega1284P

Most of these are listed in the `--devices` command, although ATmega164A isn't.

aerobaticant, 2021-03-03:
@Ro5bert would you like me to add the missing includes (from Atmel Studio) in a pull request?
I'm not that familiar with git although I've been using svn for years. :-)

Ro5bert, 2021-03-03:
@aerobaticant Thanks for the issue. I would've just let you make the PR you suggested, but I've been meaning to pillage all the include files from Atmel Studio (I don't use windows) so that I can easy add them whenever new devices are added to AVRA, and this gave me a reason to do it. I added all the include files you mentioned, plus the remaining variants that came with Atmel Studio.

commit 7a6ca91b (2021-03-03): Add more inc files and change CRLF -> LF

Closes #29
closed 2021-03-03 by Ro5bert