Ro5bert/avra #34: Supporting the ATtiny1614

supakeen, 2021-05-08:
Hey there, after suffering from a severe case of NIH I've decided to write some assembly for the ATtiny1614: https://www.microchip.com/wwwproducts/en/ATTINY1614

It's not directly listed in avra's devices list, if I wanted to make a PR to support it is this a case of matching https://github.com/Ro5bert/avra/blob/master/src/device.c#L51 to the specification sheet and submitting it or is there some deeper knowledge I must have?

bwack, 2021-05-08:
If I'm not mistaken, "tinyAVR® 1-series" are not implemented. I wonder how different this core is compared to AVR and how much work it would be to implement them into AVRA.
I ran into a similar problem as you last year. I programmed an attiny85 with AVRA, and I figured that attiny drew too much power in sleep mode and I went for an attiny412. I just went with C code and avrgcc instead.

Ro5bert, 2021-05-08:
@bwack is right: the tiny1614 has what the [AVR Instruction Set Manual](http://ww1.microchip.com/downloads/en/DeviceDoc/AVR-Instruction-Set-Manual-DS40002198A.pdf) calls the AVRxt core, and no devices with this core are currently supported in AVRA. However, all the cores are machine code-compatible (except for AVRrc), so I don't think it would be much work to support the tiny1614 (and similar devices). Apparently, the main difference between AVRe+ (the core of many devices that AVRA supports) and AVRxt is instruction timing (e.g., stores are only a single cycle), but an assembler doesn't care about this. The 1-series also seems to have different peripherals and memory map, but that also shouldn't matter. The AVRxt core generally supports the `SPM Z+` instruction, which doesn't seem to be supported in [mnemonic.c](https://github.com/Ro5bert/avra/blob/master/src/mnemonic.c), but the specific device you mention doesn't support `SPM Z+` anyway.

Hence, I think you should be able to get it to work simply by adding a line to the device array, but I'm not sure. If you wanted to give it a try and post your results, that be great. For determining the proper flags of a new device, appendix A of that instruction set manual I linked is a good resource. You should find the needed tiny1614-specifc memory information on pages 19 and 20 of the datasheet you linked.

supakeen, 2021-05-08:
I'll give it a try, I've got a bunch of them lying around and a programmer. Thanks for the pointers!

ColonelPhantom, 2021-07-30:
Did you manage to get it working? I am also looking into maybe working with AVRxt devices (but then the ATMega's), and I would like to know if it can work with avra.

Ro5bert, 2021-07-30:
@ColonelPhantom It looks like someone got the ATmega4809 (an AVRxt device) working over at #40.

fritzlb, 2025-03-27:
If you're comfortable specifying registers by yourself (or copy the according file out of a microchip studio installation) it should work. I just got a tiny program working on the similar tiny1616. There are some differences in comparison to older AVRs regarding registers (like CPU.CCP and memory mapping of things), but nothing not explained in the datasheet. Else its pretty similar I'd say