Ro5bert/avra #47: Unexpected error Branch out of range (-64 <= k <= 63)

w5277c, 2022-05-07:
Hi,

In projects with a lot of labels, I get an unexpected error:
core5277/./core/ram/_ram_find_used_block_mt.inc(42) : Error   : Branch out of range (-64 <= k <= 63)

In code:
	LDI LOOP_CNTR,0x00
	LDI ZH,high( _C5_DRIVERS_HEADER)
	LDI ZL,low( _C5_DRIVERS_HEADER)
_C5_RAM_FIND_USED_BLOCK_MT__LOOP1:
	CALL _C5_RAM_FIND_USED_BLOCK_MT__BODY
	ADIW ZL,_C5_DRIVER_HEADER_SIZE
	INC LOOP_CNTR
	CPI LOOP_CNTR,C5_DRIVERS_QNT
;42 line:
	BRNE _C5_RAM_FIND_USED_BLOCK_MT__LOOP1

In other projects, the error appears in other places, but is always associated with labels and branches.

Maybe the assembler doesn't compare labels correctly?

Reproducible on current version as well as on 1.3.0

Best regards,
Konstantin.

w5277c, 2022-05-07:
This code also doesn't work:

_C5_RAM_FIND_USED_BLOCK_MT__LOOP1:
	BRNE _C5_RAM_FIND_USED_BLOCK_MT__LOOP1

w5277c, 2022-05-07:
List file:
          _C5_RAM_FIND_USED_BLOCK_MT__LOOP1:
C:00053d   +  <>MCALL _C5_RAM_FIND_USED_BLOCK_MT__BODY
C:00053d 940e 053f CALL _C5_RAM_FIND_USED_BLOCK_MT__BODY
C:00053f 9636      <--->ADIW ZL,_C5_DRIVER_HEADER_SIZE
C:000540 9553      <--->INC LOOP_CNTR
C:000541 3150      <--->CPI LOOP_CNTR,C5_DRIVERS_QNT
C:000542 f6e1      <--->BRNE _C5_RAM_FIND_USED_BLOCK_MT__LOOP1

w5277c, 2022-05-16:
This issue is no longer actual for me.
Although the work is not yet fully completed.

### **Thank you very much, AVRA sources is very helpful.**

avra -I /home/kostas/repos/w5277c/core5277/  "main.asm" -m "main.asm".map -l "main.asm".lst (в каталоге: /home/kostas/repos/w5277c/5277.ru/firmware/solid_relay_x4_v1.0)
...
../common/relay.inc(374) : Error   : Branch out of range (-64 <= k <= 63)
../common/history.inc(34) : Error   : Branch out of range (-64 <= k <= 63)
../common/history.inc(102) : Error   : Branch out of range (-64 <= k <= 63)
../common/history.inc(112) : Error   : Branch out of range (-64 <= k <= 63)
../common/master.asm(365) : Error   : Branch out of range (-64 <= k <= 63)
../common/master.asm(422) : Error   : Branch out of range (-64 <= k <= 63)
../common/master.asm(432) : Error   : Branch out of range (-64 <= k <= 63)
../common/master.asm(466) : Error   : Branch out of range (-64 <= k <= 63)
../common/master.asm(476) : Error   : Branch out of range (-64 <= k <= 63)
../common/master.asm(506) : Error   : Branch out of range (-64 <= k <= 63)
../common/master.asm(506) : Maximum error count reached. Exiting...
main.asm(136) : Maximum error count reached. Exiting...
AVRA: advanced AVR macro assembler Version 1.3.0 Build 1 (8 May 2010)
Copyright (C) 1998-2010. Check out README file for more info
   AVRA is an open source assembler for Atmel AVR microcontroller family
   It can be used as a replacement of 'AVRASM32.EXE' the original assembler
   shipped with AVR Studio. We do not guarantee full compatibility for avra.
   AVRA comes with NO WARRANTY, to the extent permitted by law.
   You may redistribute copies of avra under the terms
   of the GNU General Public License.
   For more information about these matters, see the files named COPYING.
Pass 1...
Pass 2...
done
Used memory blocks:
   Code      :  Start = 0x0000, End = 0x0048, Length = 0x0049
   Code      :  Start = 0x1E00, End = 0x1FE2, Length = 0x01E3
   Code      :  Start = 0x0049, End = 0x1650, Length = 0x1608
Assembly aborted with 10 errors and 2 warnings.

-And----------------------------------------------------------------------------------

JAVRA Java AVR macro assembler Version 0.0.1
Licensed by GPL-3.0-or-later

WARNING! The project is not finished yet, it is under development.
!!!It is not recommended to use!!!

Warning[1] ./devices/atmega168.inc(18): unsupported #pragma
Warning[2] ./devices/atmega168.inc(19): unsupported #pragma
Warning[3] ./devices/atmega168.inc(20): unsupported #pragma
Warning[4] ./devices/atmega168.inc(21): unsupported #pragma
Message: ######## device atmega168
Warning[5] ./boot/bus5277_bldr.inc(47): err_cntr(r23) already assigned
Warning[6] ./boot/bus5277_bldr.inc(48): bus_addr(r25) already assigned
Message: ######## buffer offset:584
Message: ######## buffer size:168
Message: ######## available ram:459
Message: ######## drivers headers offset:384
Message: ######## tasks headers offset:480
Message: ######## free ram offset:752
Message: ######## stack end offset:1215
Message: ######## core freq:16
Message: ######## timers normal speed (100)
Message: ######## timers x2 speed (100)
Message: ######## io baudrate:230400
Message: ######## logging disabled
Message: ######## dispatcher enabled
Message: ######## timers enabled
Message: included driver hardware uart v0.6
Message: included driver bus5277 v0.4
Message: included driver buttons v0.4
Message: included driver 1wire v0.3
Message: included driver ds18b20 v0.8
Message: included driver hardware pcint v0.2
Message: included driver am2301 v0.6
--CODE-----------------------------------
 Start	= 0000, End = 0048, Length = 0049
 Start	= 1E00, End = 1FE2, Length = 01E3
 Start	= 0049, End = 1650, Length = 1608

 Total	:  6196 words (12392 bytes)

Build SUCCESS, warnings:6
(parsed: 7472 lines, total time: 0.45 s)

Ro5bert, 2022-05-30:
Sorry I was of no help, but I'm glad this isn't a problem for you any more. And thanks for pointing out some bugs in mnemonics.c, although I don't have time to look into it them right now. IMO, most of mnenomics.c is a mess and should be rewritten.