Ro5bert/avra #53: Expressions should be evaluated as 64-bit integer

srtlg, 2022-09-03:
For compatibility with AVRASM2 all expressions should be 64 bit internally, so that the following is equal

```assembly
.device atmega168
.equ F_CPU = 16000000
.macro delay1
        .set cycles = @0 * (F_CPU / 1000000)
        .dw cycles
.endmacro
.macro delay2
        .set cycles = (@0 * F_CPU) / 1000000
        .dw cycles
.endmacro
delay1 1000
delay2 1000
```

srtlg, 2022-09-03:
here is the relevant section of the manual
[image]

imrehorvath, 2023-02-06:
I just run into the same issue with a similar problem, using macros to calculate the cycles needed for delays. It would be nice to have 64 bits integers for the values in expressions as @srtlg  suggested.